adafruit/Adafruit_CAD_Parts #1: Part Requests

McCallisterMicro, 2018-08-26:
Comments on the article from July linking to this are no longer available, but I wanted to put in a request for a few parts I use regularly:
#815 16-Channel 12-bit PWM/Servo Driver - I2C interface - PCA9685
#1429 24-Channel 12-bit PWM LED Driver - SPI Interface - TLC5947
#1455 12-Channel 16-bit PWM LED Driver - SPI Interface - TLC59711
#1602 CAP1188 - 8-Key Capacitive Touch Sensor Breakout - I2C or SPI
#1982 Adafruit 12-Key Capacitive Touch Sensor Breakout - MPR121
#395 8-channel Bi-directional Logic Level Converter - TXB0108
#2190 VERTER 5V USB Buck-Boost - 500mA from 3V-5V / 1000ma from 5V-12V

Alternatively, I may be working in the meantime to do the conversions myself in Eagle 9, but that can be quite tedious when all the parts need to be copied into managed libraries for 3d models. 

Thanks!

McCallisterMicro, 2018-09-12:
Ah! totally missed those. I see that one of the TLCs have made it on too. Thanks!

nineseventytwo, 2019-08-09:
#2471 - Adafruit HUZZAH ESP8266 Breakout.

djecken, 2020-03-20:
@nineseventytwo just added part, cheers! https://github.com/adafruit/Adafruit_CAD_Parts/tree/master/2471%20HUZZAH%20ESP8266%20Breakout

h3llb3nt, 2021-02-09:
https://www.adafruit.com/product/1768
the neopixel 1/4 60 ring segments?