adafruit/Adafruit_CAD_Parts #15: More Parts Request

whittedb, 2019-05-24:
Metro Mini
All form factors of the FX sound board

AGenews, 2019-07-29:
**3321 Mini Color TFT with Joystick FeatherWing** .... please & Thanks a lot!
Best,
Andreas

djecken, 2019-10-31:
@AGenews just added 3321 Mini color TFT with Joystick, cheers!

AGenews, 2019-10-31:
WOW THANKS A LOT!!!

djecken, 2020-03-20:
Just added 3 audio FX sound boards, cheers! https://github.com/adafruit/Adafruit_CAD_Parts/tree/master/2133%20Audio%20FX%20Soundboard%20with%20Headphone https://github.com/adafruit/Adafruit_CAD_Parts/tree/master/2217%20Adafruit%20FX%20Soundboard%20with%20Amp https://github.com/adafruit/Adafruit_CAD_Parts/tree/master/2341%20Adafruit%20EZ-SFX%20Mini

djecken, 2020-03-20:
metro mini just added https://github.com/adafruit/Adafruit_CAD_Parts/tree/master/2590%20Metro%20Mini

closed 2020-03-20 by djecken

whittedb, 2020-03-22:
Awesome!  Thank you.