adafruit/Adafruit_CAD_Parts #24: Model Request: 4022 - Triple-axis Magnetometer - MLX90393

mscosti, 2020-01-31:
Since a typical use case of the MLX90393 is for position/orientation sensing of magnets, or rotary encoding, its important to get the alignment of the board and magnet just right. A model file for it would be great when designing something to house this sensor + magnets.

Thanks

dandelany, 2020-02-11:
Hi! I've been learning how to make 3d models from Eagle files, so I modeled your request for practice :) I submitted a pull request here: #26 

In the meantime, you can [get it from my branch if you need it](https://github.com/dandelany/Adafruit_CAD_Parts/tree/mlx90393/4022%20Wide-Range%20Triple-axis%20Magnetometer%20MLX90393). Hope it's useful :)

mscosti, 2020-02-13:
That's awesome, thank's so much! I See it's already been merged, which is even better. I'll Pull that down soon and try it out :)

Out of curiosity, are there any particular resources you were using to learn how to do this? I would be interested in checking them out to know the general process, if I ever wanted to try this out myself.

mscosti, 2020-02-14:
Works and looks great! Thanks a bunch! 

Closing, #26 resolves.

closed 2020-02-14 by mscosti

dandelany, 2020-02-20:
No problem! I mostly used [this tutorial video](https://www.youtube.com/watch?v=S8U09Pj1m6M) from Adafruit.