adafruit/Adafruit_CAD_Parts #142: Incorrect import? 2809 LIS3DH has incorrect dimensions and features, no components

nicolasff, 2023-06-27:
Hello,

I imported the model for the LIS3DH accelerometer breakout board today, trying both the Fusion 360 file (`.f3d`) and the STEP file provided in https://github.com/adafruit/Adafruit_CAD_Parts/tree/ea4f501efb3197b5fa934d310145f21a79ef87cc/2809%20LIS3DH.

The 3D model does not match the physical product, which made me wonder if these files were maybe for a different product and ended up in this directory by mistake. The git history for `2809 LIS3DH.f3d` shows that it was [edited and moved](https://github.com/adafruit/Adafruit_CAD_Parts/commit/a13f14591e1ffddc5d5d0e0ac00b45189531eb14#diff-1ebf544f621b8139efe8cfdb8fd8e5d87cc3c015c81fea32690c37fefabba637) into its current directory with commit a13f14591e1ffddc5d5d0e0ac00b45189531eb14 ~5 years ago. I also opened the version of the `.f3d` file that was [pre-edit/pre-move](https://github.com/adafruit/Adafruit_CAD_Parts/blob/b0f9d0c57f3735973f604701bc06a5cefcb8c1b1/2809%20LIS3DH.f3d), but it still had the same contents.

→ Is there an accurate 3D model of this board available somewhere?

### Visual comparison

Here is what the breakout board looks like (from the product page):

[image]

and this is what both the Fusion 360 and STEP files contain (shown shaded with hidden edges):

[image]

There is only a single "body" in the `.f3d` file, unlike [some other files](https://github.com/adafruit/Adafruit_CAD_Parts/blob/ea4f501efb3197b5fa934d310145f21a79ef87cc/4479%20LIS3MDL/4479%20LIS3MDL.jpg) which have different components on the PCB modeled separately.

### Dimensions

The dimensions of the 3D object are 20.32×20.32×1.60 mm, with the larger holes having a diameter of 2.54mm (1/10") and the smaller ones 0.889mm (35 × 1/1000th of an inch).

I measured the breakout board with calipers at 25.59×17.79×1.60 mm.

djecken, 2023-06-27:
Hello, thanks for reporting this issue. The models for the LIS3DH have been updated to the latest revision of the breakout board https://github.com/adafruit/Adafruit_CAD_Parts/tree/main/2809%20LIS3DH

closed 2023-06-27 by djecken

nicolasff, 2023-06-27:
Thank you! That was so quick.
The new model looks great with lots of details, and has the dimensions I expected.