adafruit/Adafruit_CAD_Parts #158: Can't clone

tultalk1, 2023-10-24:
Cloning into 'C:\Users\15172\Source\Repos\4538%20NAU7802'...
Error encountered while cloning the remote repository: Git failed with a fatal error.
repository 'https://github.com/adafruit/Adafruit_CAD_Parts/tree/main/4538%20NAU7802/' not found

todbot, 2023-10-24:
Looks like whatever program you're using to do the clone can't handle spaces.
What are you using?

caternuson, 2023-10-24:
Need to clone the entire repo, not a subfolder.
```
git clone https://github.com/adafruit/Adafruit_CAD_Parts.git
```

Can also download entire repo as zip.

Single files can be downloaded once navigated to:
[image]

caternuson, 2023-10-25:
Closing. Not a repo issue.

closed 2023-10-25 by caternuson