adafruit/Adafruit_CAD_Parts #187: 2090 2.8in TFT Cap Touch - All 4 files seem to be corrupted

frodo42, 2024-08-10:
I've tried importing the files into Onshape, Fusion 360, meshmixer, and Cura (where appropriate by file type). Each time I get an error (translation error or corrupted file error). Can you please check the uploaded files? Thanks!

todbot, 2024-08-10:
The STL and JPG seem to render okay on github:
- https://github.com/adafruit/Adafruit_CAD_Parts/blob/main/2090%202.8in%20TFT%20Cap%20Touch/2090%202.8in%20TFT%20Cap%20Touch.stl
- https://github.com/adafruit/Adafruit_CAD_Parts/blob/main/2090%202.8in%20TFT%20Cap%20Touch/2090%202.8in%20TFT%20Cap%20Touch.jpg

What's your process for getting these files and importing them?

frodo42, 2024-08-10:
Just did a fresh download (used the download raw) from before and the import worked into Onshape perfectly this time. No idea what happened, but must have been on the download side on my end. Thanks for looking into this for me though!

djecken, 2024-08-12:
@todbot Thanks for your help! Closing out this issue

closed 2024-08-12 by djecken